ClockVapor/rpi_gpio #2: Trying to run this gem within Docker, however it returns the error 'RuntimeError: this gem can only be run on a Raspberry Pi'

jrobertson, 2015-08-16:
I'm running the latest hypriot version of Raspbian (Jessie) which is running Ruby (2.2.0) inside a Docker container and I have installed the gem *rpi_gpio*. When I open an irb session and after entering `require 'rpi_gpio'`, it returns the error message <pre>RuntimeError: this gem can only be run on a Raspberry Pi </pre>

Can you fix it to run without a problem inside Docker on the Raspberry Pi?

ClockVapor, 2015-08-17:
Sadly I don't have any experience working with Docker. However, that error may be an incompatibility with your RPi hardware. Which model are you using?

As far as I can tell, the gem currently doesn't work on RPi 2 models, but I plan to upgrade it so it's up to par with the latest RPi.GPIO Python module, which does work on RPi 2.

jrobertson, 2015-08-17:
I'm closing this issue because I now believe it is a RPi 2 compatibility issue. I read somewhere yesterday evening that there was a problem with reading the /proc/cpuinfo in RPi 2 from an existing C library.

closed 2015-08-17 by jrobertson

ClockVapor, 2015-08-18:
I just updated the gem to support the latest RPi.GPIO Python module version, so it should work on all RPi models. Give it a shot! And thank you again for letting me know that it wasn't working.