ClockVapor/rpi_gpio #5: Channels switch on(low) when setting to output.

iohzrd, 2016-08-04:
I don't know if this is a bug or something wrong with my setup but at the beginning of the script I'm working on I initialize like so:

RPi::GPIO.set_numbering :board
pin = {1=>5,2=>7,3=>8,4=>10,5=>11,6=>12,7=>13,8=>15,9=>16,10=>18,11=>19,12=>21,13=>22,14=>23,15=>24,16=>26}
for i in 1..16
    RPi::GPIO.setup pin[i], :as => :output
    RPi::GPIO.set_high pin[i]
end

Yet all the channels switch on, regardless of the set_high directive.
Any thoughts?

ClockVapor, 2016-08-04:
Nope, looks like there isn't an "initial" argument... I must've missed that one in the conversion. Thanks for the notice!

iohzrd, 2016-08-04:
My pleasure, I'm taking a crack at implementing it myself but don't let that stop you from doing so cause I'm new to ruby c extensions. Thanks for the reply and the good work.

iohzrd, 2016-08-05:
I got it working, I'll commit and do a pull request soon so you can review it, It might have bugs.

ClockVapor, 2016-08-13:
I merged your pull request. Thanks for the contribution!

closed 2016-08-13 by ClockVapor