ClockVapor/rpi_gpio #7: wrong commit author

iohzrd, 2016-08-22:
Hey man, I hate to bug you with this but my commit/pull request was submitted with the wrong git author/email. I know it's inconvenient but I would greatly appreciate if you would roll back to before my pull request so I can create a new one with my updated settings. I have already fixed it on my end.

ClockVapor, 2016-08-31:
Sorry for the delay. And no worries, I'll roll it back sometime tomorrow.

iohzrd, 2016-09-04:
Thanks bro, no rush, I appreciate it.

ClockVapor, 2016-09-10:
And by "tomorrow," I meant "in 11 days." :stuck_out_tongue:

It's rolled back now.

ClockVapor, 2016-10-08:
Just noticed that your PR didn't have you as an author again for some reason. Let me know if you want to try again so I can roll back before you make your changes on your fork,

closed 2016-11-23 by ClockVapor