ClockVapor/rpi_gpio #19: Doesn't work properly with more than 2 LEDs

Souravgoswami, 2019-07-27:
Hi, I created a script that blinks the LEDs on channel 12, 7, and 8.
I am using Arch Linux ARM on Rapsberry Pi 3 model B.
My code:
```
#!/usr/bin/ruby -w
require 'rpi_gpio'
Object.prepend(RPi::GPIO)

set_warnings(false)
set_numbering(:board)

pins = [12, 7, 8].each { |x| setup(x, as: :output) }
c = pins.size.times.map { |x| ?- * x.next }

begin
	loop while pins.each_with_index do |x, i|
		print "\e[2KPin #{c[i]}> #{x}\r"
		set_high(x)
		Kernel.sleep(0.5)
		set_low(x)
		Kernel.sleep(0.5)
	end

rescue SignalException, Interrupt, SystemExit
	pins.each(&method(:clean_up))
	exit 0

rescue Exception => e
	Kernel.warn(e)
	Kernel.sleep 1
	retry
end
```
I see irregular flashes. The LED [at pin] 12 turns on and off first. Then LED 7, then LED 8 without turning LED 7 off. Then LED 12 goes on and off. Then 7 goes off and on after a while. Then LED 8 goes off and on and it continues. It looks messy. This shouldn't happen! Same happens with PWM...

Souravgoswami, 2019-07-27:
Not sure why. set_high and set_low reverses for pin 7, 8 and 10, 12.
That is if set_high turns off the light on pin 7 or 8, it will turn on the light on 10 or 12...
Why is this the case?

jezcaudle, 2020-05-14:
I've changed your code slightly:

`set_numbering(:bcm)`
`pins = [4, 17, 18, 27,22,23,24,5,6,12,13].each { |x| setup(x, as: :output) }` 

and it works perfectly on my Pi Zero. All the lights come on, then all off and then each one comes on and then off. I use BCM because the development board has BCM numbers etched on them.

`pins = [4, 17, 18, 27,22,23,24,5,6,12,13].each { |x| setup(x, as: :output, initialize: :low) }`

and the lights start off.

I'm using a Pi Zero, Buster and ruby 2.7.0p0 (2019-12-25 revision 647ee6f091) [armv6l-linux-eabihf] installed using RVM, rpi_gpio (0.3.3).

I've left it running for 30 minutes and it still works as it should. 

Might it be down to something else using the serial RXD and TXD - pins 8 and 10? Pin 7 is also the General Purpose Clock. 

If you move the LED's to different, GPIO only pins, do you get the same problems? For example pins 11, 13, 15 and 16?

Also does your Pi have enough amps? I've had problems with that in the past.

ameuret, 2021-11-09:
So, @Souravgoswami ?

Souravgoswami, 2021-11-10:
Not really exactly sure, I don't have the raspberry pi setup any more :(

Maybe it can be closed?