ClockVapor/rpi_gpio #22: Can't Require Gem on RPi 4

aaronmiler, 2020-08-19:
I get a load error on a Raspberry Pi 4B for version `0.5.0`. Version `0.4.0` loads without issue, here's the stack trace

I see the Readme mentions Ruby 2.0.0 as the highest version tested, but on the Pi4, 2.4 is the lowest version I could easily install with `rbenv`. 

```
irb(main):001:0> require 'rpi_gpio'
LoadError: libruby.so.2.7: cannot open shared object file: No such file or directory - /home/pi/.rbenv/versions/2.4.0/lib/ruby/gems/2.4.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio/rpi_gpio.so
	from /home/pi/.rbenv/versions/2.4.0/lib/ruby/gems/2.4.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio.rb:1:in `require'
	from /home/pi/.rbenv/versions/2.4.0/lib/ruby/gems/2.4.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio.rb:1:in `<top (required)>'
	from (irb):1:in `require'
	from (irb):1
        [ bundler stack trace continues ]
```

`lib/rpi_gpio/rpi_gpio.so` does exist on my machine. I also tried using the code from #20, but get the `RuntimeError: this gem can only be run on a Raspberry Pi`. 

Happy to perform any debugging steps on my machine

mikekreuzer, 2020-09-08:
Same error, Pi Model 2B v1.1, Ruby 2.5 -- 2.5 being what I got with apt-get install ruby-full, and even that's in "security maintenance phase (will EOL soon!)"

jamoore5, 2020-09-13:
Same on on Pi Model 3B+ with ruby 2.6.6

```
pi@raspberrypi:~ $ irb
irb(main):001:0> require 'rpi_gpio'
Traceback (most recent call last):
       10: from /home/pi/.rbenv/versions/2.6.6/bin/irb:23:in `<main>'
        9: from /home/pi/.rbenv/versions/2.6.6/bin/irb:23:in `load'
        8: from /home/pi/.rbenv/versions/2.6.6/lib/ruby/gems/2.6.0/gems/irb-1.0.0/exe/irb:11:in `<top (required)>'
        7: from (irb):1
        6: from /home/pi/.rbenv/versions/2.6.6/lib/ruby/2.6.0/rubygems/core_ext/kernel_require.rb:34:in `require'
        5: from /home/pi/.rbenv/versions/2.6.6/lib/ruby/2.6.0/rubygems/core_ext/kernel_require.rb:130:in `rescue in require'
        4: from /home/pi/.rbenv/versions/2.6.6/lib/ruby/2.6.0/rubygems/core_ext/kernel_require.rb:130:in `require'
        3: from /home/pi/.rbenv/versions/2.6.6/lib/ruby/gems/2.6.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio.rb:1:in `<top (required)>'
        2: from /home/pi/.rbenv/versions/2.6.6/lib/ruby/2.6.0/rubygems/core_ext/kernel_require.rb:54:in `require'
        1: from /home/pi/.rbenv/versions/2.6.6/lib/ruby/2.6.0/rubygems/core_ext/kernel_require.rb:54:in `require'
LoadError (libruby.so.2.7: cannot open shared object file: No such file or directory - /home/pi/.rbenv/versions/2.6.6/lib/ruby/gems/2.6.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio/rpi_gpio.so)
```

innersund, 2020-11-09:
I faced a similar LoadError complaining "No such file ..." although the file clearly existed. I was using the latest Ruby from the repo, which is 2.5.0, and since the LoadError indicated libruby.so.2.7 I guessed it might be a version problem. So I followed a guide to install latest version of Ruby on Raspberry pi :
`sudo apt install rbenv
git clone https://github.com/rbenv/ruby-build.git ~/.rbenv/plugins/ruby-build
rbenv install 2.7.2
PATH=~/.rbenv/versions/2.7.2/bin:$PATH
gem install rpi_gpio`
and the error went away.

innersund, 2020-11-09:
I faced a similar LoadError complaining "No such file ..." although the file clearly existed. I was using the latest Ruby from the repo, which is 2.5.0, and since the LoadError indicated libruby.so.2.7 I guessed it might be a version problem. So I followed a guide to install latest version of Ruby on Raspberry pi :
```
sudo apt install rbenv
git clone https://github.com/rbenv/ruby-build.git ~/.rbenv/plugins/ruby-build
rbenv install 2.7.2
PATH=~/.rbenv/versions/2.7.2/bin:$PATH
gem install rpi_gpio
```
and the error disappeared.

mbegley, 2020-11-30:
I am seeing the same issue, and have a janky workaround.  I'm going to try to improve on it but I'm putting it out there in case it might help someone immediately, and to get the conversation started.

Essentially, c_gpio.c, setup() sets up the memory map between the GPIO device and a block of memory.  The function tries to do this in two ways, the first if it has permissions to read from /dev/gpiomem, and the second if it does not.

I haven't been able to get get the first mapping working yet, because I don't have permission as a user to read from this device.  I think I need to give ruby setuid root permissions to do this, and I haven't taken that step yet.  Instead, I've gotten the second code path to work, which does require ruby to be run as root.

In this case, the IO memory base address, or peri_base, isn't getting determined properly by setup.c - the code is evaluating it to be zero and the code is bailing with this call:

   if (!peri_base)
      return SETUP_NOT_RPI_FAIL;

peri_base on the raspberry pi 4 needs to be 0xFE000000.  So in the #define's I set this based on the CPU type:

#define BCM2711_PERI_BASE_DEFAULT   0xFE000000

Later on, there are two attempts to set peri_base, again, based on permissions.  In the first attempt, it attempts to read from /proc/device-tree/soc/ranges and computes the value based on the contents of that device file.  I don't yet understand the computation, so for now I've simply replaced the line that computes peri_base with one that sets peri_base to this value:
peri_base = BCM2711_PERI_BASE_DEFAULT;

In the second attempt, which tries to determine peri_base based on the cpu type by reading /proc/cpuinfo, I've added an elseif based on the CPU type BCM2711:

    else if (strcmp(hardware, "BCM2711")) {
        // pi 4 hardware
        peri_base = BCM2711_PERI_BASE_DEFAULT;
        found = 1;

Now, if I build this code and copy the resulting shared library into place, and run my test code as root, I'm able to read and write to the GPIO pins.

Anyway, here's my modified c_gpio.c for you to copy into your source tree.  I doubt it'll work on anything but a pi 4 right now, but it's a start.
---
/*
Original code by Ben Croston modified for Ruby by Nick Lowery
(github.com/clockvapor)
Copyright (c) 2014-2020 Nick Lowery

Copyright (c) 2012-2019 Ben Croston

Permission is hereby granted, free of charge, to any person obtaining a copy of
this software and associated documentation files (the "Software"), to deal in
the Software without restriction, including without limitation the rights to
use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies
of the Software, and to permit persons to whom the Software is furnished to do
so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in all
copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
SOFTWARE.
*/

#include <stdio.h>
#include <stdint.h>
#include <stdlib.h>
#include <fcntl.h>
#include <sys/mman.h>
#include <string.h>
#include "c_gpio.h"

#define BCM2708_PERI_BASE_DEFAULT   0x20000000
#define BCM2709_PERI_BASE_DEFAULT   0x3f000000
#define BCM2711_PERI_BASE_DEFAULT   0xFE000000
#define GPIO_BASE_OFFSET            0x200000
#define FSEL_OFFSET                 0   // 0x0000
#define SET_OFFSET                  7   // 0x001c / 4
#define CLR_OFFSET                  10  // 0x0028 / 4
[... 245 lines cut ...]

    if (value) // value == HIGH
        offset = SET_OFFSET + (gpio/32);
    else       // value == LOW
       offset = CLR_OFFSET + (gpio/32);

    shift = (gpio%32);

    *(gpio_map+offset) = 1 << shift;
}

int input_gpio(int gpio)
{
   int offset, value, mask;

   offset = PINLEVEL_OFFSET + (gpio/32);
   mask = (1 << gpio%32);
   value = *(gpio_map+offset) & mask;
   return value;
}

void cleanup(void)
{
    munmap((void *)gpio_map, BLOCK_SIZE);
}

mbegley, 2020-11-30:
With the updated code in my comment above, I'm able to toggle an LED with this bit of ruby:
```
require 'rpi_gpio'

RPi::GPIO.set_numbering :board
RPi::GPIO.setup 11, :as => :output

while true do
  RPi::GPIO.set_low 11
  sleep 1
  RPi::GPIO.set_high 11
  sleep 1
end
```

repent, 2021-02-20:
Same issue on a Pi 3B, with Ruby 3.0.0 and 2.5.1.  Before finding this thread I'd tried rbenv, RVM and a clean install of Raspbian.

mbegley, 2021-02-20:
> Any update?
> I have exactly same issue on pi4 and I don't know how to rebuild the gem locally with the updated c_gpio.c.txt. Thanks @mbegley

These may not be comprehensive instructions, but they may be enough to muddle through:

1) clone down a copy of the rpi_gpio github repo
2) in the repo directory tree, navigate to ./ext/rpi_gpio/
3) in that directory, make a copy of the file c_gpio.c so you can easily roll back.
4) edit g_gpio using your favorite editor, as described above.  Assuming the code hasn't been updated since I made this port, youcan likely substitute in the contents of c_gpio.c.text that's attached to my earlier comment.
5) build the shared library my typing: make
6) If the build completes successfully, then it will have created a new shared library: 'rpi_gpio.so'. You should be able to copy the new shared library into place so your system can use it.  Be sure to make a copy of the original shared library before you move it, again, for easier rollback if there are problems.

I updated it in two places:

/usr/local/lib/ruby/gems/2.7.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio
and 
~/.rvm/gems/ruby-2.7.2/gems/rpi_gpio-0.5.0/lib/rpi_gpio/

Now, check to see if your system works and you're able to talk to the GPIO!

Anyway, this is the steps I followed a couple months ago, based on memory.  I hope it's clear (and accurate) enough - let me know if you have any issues or need clarifications.

Obviously, this should only be done on an as-needed basis, and should be considered a particularly ugly bandaid.  I hope it can help someone out until a proper fix can be rolled out.

repent, 2021-02-24:
> @mbegley "I haven't been able to get get the first mapping working yet, because I don't have permission as a user to read from this device. I think I need to give ruby setuid root permissions to do this, and I haven't taken that step yet. Instead, I've gotten the second code path to work, which does require ruby to be run as root."

Sorry if this is a dumb suggestion, but your user is a member of the gpio group, right?

mbegley, 2021-02-24:
Not a dumb suggestion, because the user wasn't a member of the group.  I just added it, and I still get an error unless I'm running as root.  :-/

I'm probably doing something dumb.

rojosinalma, 2021-03-11:
The problem itself is that the library Ruby expects should be in `/lib/rpi_gpio/rpi_gpio.so`.

This file is created at instalation time (or when manual compilation happens), for some reason, your compilation process either has an error or the file is just not being created, that's why Ruby then complains that it can't find it.

I'm not sure the solution is indeed updating the start values for any given BCM, but the fact that your current BCM (i.e: Pi4) is not supported, but I very much doubt that's the problem.

I haven't tried the instructions here, but I did checkout the only open PR right now (#20), compiled the lib and required this gem directly from my checkout instead of the one installed by rubygems. 

If all of this sounds weird to you, the tl;dr is: 

Try recompiling the library:
_you need to manually install rake and rake-compiler_

```
rake clean
rake compile
```

This process should output a fiile: `/lib/rpi_gpio/rpi_gpio.so`
If this file doesn't exist, then you need to debug why the compilation fails, that I leave to you.

If you solve that or manage by any other means to get that library there, this problem should be gone.