ClockVapor/rpi_gpio #24: rpi_gpio.so compiled as 32-bit on 64-bit system

vanne, 2020-09-22:
On Ubuntu 20.04 LTS 64-bit, the gem installs fine but rpi_gpio.so is compiled as a 32-bit lib.

"rpi_gpio.so: wrong ELF class: ELFCLASS32"

```
ubuntu@rpi4:~/test$ uname -a
Linux rpi4 5.4.0-1018-raspi #20-Ubuntu SMP Sun Sep 6 05:11:16 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
ubuntu@rpi4:~/test$ rbenv version
2.7.1 (set by /home/ubuntu/.rbenv/version)
ubuntu@rpi4:~/test$ 

ubuntu@rpi4:~/test$ cat test.rb 
require 'rpi_gpio'
ubuntu@rpi4:~/test$ bundle exec ruby test.rb 
Traceback (most recent call last):
	3: from test.rb:1:in `<main>'
	2: from test.rb:1:in `require'
	1: from /home/ubuntu/.rbenv/versions/2.7.1/lib/ruby/gems/2.7.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio.rb:1:in `<top (required)>'
/home/ubuntu/.rbenv/versions/2.7.1/lib/ruby/gems/2.7.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio.rb:1:in `require': /home/ubuntu/.rbenv/versions/2.7.1/lib/ruby/gems/2.7.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio/rpi_gpio.so: wrong ELF class: ELFCLASS32 - /home/ubuntu/.rbenv/versions/2.7.1/lib/ruby/gems/2.7.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio/rpi_gpio.so (LoadError)
ubuntu@rpi4:~/test$
```

vanne, 2020-09-22:
When checking out the repository and building manually, things are fine:

```
ubuntu@rpi4:~$ git clone https://github.com/ClockVapor/rpi_gpio.git
Cloning into 'rpi_gpio'...
remote: Enumerating objects: 110, done.
remote: Counting objects: 100% (110/110), done.
remote: Compressing objects: 100% (65/65), done.
remote: Total 426 (delta 51), reused 73 (delta 27), pack-reused 316
Receiving objects: 100% (426/426), 241.55 KiB | 1.05 MiB/s, done.
Resolving deltas: 100% (242/242), done.
ubuntu@rpi4:~$ cd rpi_gpio/ext/rpi_gpio/
ubuntu@rpi4:~/rpi_gpio/ext/rpi_gpio$ ruby extconf.rb 
creating Makefile
ubuntu@rpi4:~/rpi_gpio/ext/rpi_gpio$ make
<snip>
linking shared-object rpi_gpio/rpi_gpio.so
ubuntu@rpi4:~/rpi_gpio/ext/rpi_gpio$ file rpi_gpio.so 
rpi_gpio.so: ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, BuildID[sha1]=9d7712484c2ed99db38dc4aa316ed47cc853e18a, with debug_info, not stripped
ubuntu@rpi4:~/rpi_gpio/ext/rpi_gpio$ make install
/usr/bin/install -c -m 0755 rpi_gpio.so /home/ubuntu/.rbenv/versions/2.7.1/lib/ruby/site_ruby/2.7.0/aarch64-linux/rpi_gpio
ubuntu@rpi4:~/rpi_gpio/ext/rpi_gpio$ 
```

phykos, 2020-10-26:
Encountered this error because I am on AArch64 too.
So far, on 5 years of Ruby I never seen such error.

chris-clifton, 2020-12-24:
Specifying the git repo did it for me and it correctly compiled as 64 bit.

1. Delete Gemfile.lock
2. Change Gemfile to read: `gem 'rpi_gpio', git: 'git@github.com:ClockVapor/rpi_gpio.git'`
3. Run `bundle install`