ClockVapor/rpi_gpio #25: Cannot open shared object

MarioRuiz, 2021-04-06:
Getting next error on a rpi 3B+

```
	5: from /usr/lib/ruby/2.5.0/rubygems/core_ext/kernel_require.rb:39:in `require'
	4: from /usr/lib/ruby/2.5.0/rubygems/core_ext/kernel_require.rb:135:in `rescue in require'
	3: from /usr/lib/ruby/2.5.0/rubygems/core_ext/kernel_require.rb:135:in `require'
	2: from /var/lib/gems/2.5.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio.rb:1:in `<top (required)>'
	1: from /usr/lib/ruby/2.5.0/rubygems/core_ext/kernel_require.rb:59:in `require'
/usr/lib/ruby/2.5.0/rubygems/core_ext/kernel_require.rb:59:in `require': libruby.so.2.7: cannot open shared object file: No such file or directory - /var/lib/gems/2.5.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio/rpi_gpio.so (LoadError)
```

repent, 2021-04-15:
@MarioRuiz I think this might be a duplicate of #22 

Before getting into that thread though, check that your user is a member of the gpio group.

synthead, 2021-11-28:
I'm running into the same problem:

```
[1] pry(main)> require "rpi_gpio"
LoadError: libruby.so.2.7: cannot open shared object file: No such file or directory - /home/deploy/.rbenv/versions/3.0.2/lib/ruby/gems/3.0.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio/rpi_gpio.so
from <internal:/home/deploy/.rbenv/versions/3.0.2/lib/ruby/3.0.0/rubygems/core_ext/kernel_require.rb>:85:in `require'
Caused by LoadError: cannot load such file -- rpi_gpio
from <internal:/home/deploy/.rbenv/versions/3.0.2/lib/ruby/3.0.0/rubygems/core_ext/kernel_require.rb>:85:in `require'
```

The "deploy" user has access to the gpio stuff.

martini, 2021-12-28:
I have the same issue here, rpi 3B+, ruby-2.6.6, rpi_gpio-0.5.0

```
pi@pi:~/sensor-tools $ irb 
2.6.6 :001 > require 'rpi_gpio'
Traceback (most recent call last):
       10: from /home/pi/.rvm/rubies/ruby-2.6.6/bin/irb:23:in `<main>'
        9: from /home/pi/.rvm/rubies/ruby-2.6.6/bin/irb:23:in `load'
        8: from /home/pi/.rvm/rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/gems/irb-1.0.0/exe/irb:11:in `<top (required)>'
        7: from (irb):1
        6: from /home/pi/.rvm/rubies/ruby-2.6.6/lib/ruby/site_ruby/2.6.0/rubygems/core_ext/kernel_require.rb:34:in `require'
        5: from /home/pi/.rvm/rubies/ruby-2.6.6/lib/ruby/site_ruby/2.6.0/rubygems/core_ext/kernel_require.rb:130:in `rescue in require'
        4: from /home/pi/.rvm/rubies/ruby-2.6.6/lib/ruby/site_ruby/2.6.0/rubygems/core_ext/kernel_require.rb:130:in `require'
        3: from /home/pi/.rvm/gems/ruby-2.6.6/gems/rpi_gpio-0.5.0/lib/rpi_gpio.rb:1:in `<top (required)>'
        2: from /home/pi/.rvm/rubies/ruby-2.6.6/lib/ruby/site_ruby/2.6.0/rubygems/core_ext/kernel_require.rb:54:in `require'
        1: from /home/pi/.rvm/rubies/ruby-2.6.6/lib/ruby/site_ruby/2.6.0/rubygems/core_ext/kernel_require.rb:54:in `require'
LoadError (libruby.so.2.7: cannot open shared object file: No such file or directory - /home/pi/.rvm/gems/ruby-2.6.6/gems/rpi_gpio-0.5.0/lib/rpi_gpio/rpi_gpio.so)
2.6.6 :002 > 

pi@pi:~/sensor-tools $ ls -l /home/pi/.rvm/gems/ruby-2.6.6/gems/rpi_gpio-0.5.0/lib/rpi_gpio/rpi_gpio.so
-rwxr-xr-x 1 pi pi 128252 Dec 28 00:38 /home/pi/.rvm/gems/ruby-2.6.6/gems/rpi_gpio-0.5.0/lib/rpi_gpio/rpi_gpio.so
pi@pi:~/sensor-tools $ id
uid=1000(pi) gid=1000(pi) groups=1000(pi),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),105(input),109(netdev),117(lpadmin),997(gpio),998(i2c),999(spi)
pi@pi:~/sensor-tools $ 
```

So it is no permission issue. Any other ideas?

richardpenner, 2022-05-23:
Coming into the same issue and noticed that `ldd /path/to/rpi_gpio.so` reveals a missing dependency:
```
…
libruby.so.2.7 => not found
…
```
So maybe it needs Ruby 2.7.0 installed? I'm attempting that and will report back

richardpenner, 2022-05-23:
Yeah that fixes it. For anyone else arriving here from google:
- the error message above is hard to read, but it's complaining about missing `libruby.so.2.7`, not the gpio file – that's where it experienced the error. If you run `ldd <full-path-to-rpi_gpio.so>` you'll see which dependency is missing – in my case it was `libruby.so.2.7`. I had to not just install Ruby 2.7 but actually use that version (in my case I am using `rbenv` to manage multiple Ruby installations). Interestingly I still see the missing file in `ldd` after installing 2.7, but when I set my Ruby version to 2.7.0 the gpio module works fine.

softagency, 2024-03-20:
In my case, libruby file name was different. so I created symlinks.
Here is how to fix it. 

```
# ldd /var/lib/gems/2.7.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio/rpi_gpio.so
        linux-vdso.so.1 (0x7efc8000)
        /usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76f83000)
        libruby.so.2.7 => not found
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76f14000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76dc1000)
        /lib/ld-linux-armhf.so.3 (0x76faf000)
# cd /lib/arm-linux-gnueabihf
# ln -s libruby-2.7.so libruby.so.2.7
```

eto-, 2024-07-01:
`# cd /usr/lib/arm-linux-gnueabihf & ln -s libruby-3.1.so libruby.so.2.7`

it will work