ClockVapor/rpi_gpio #27: not running on a RPi

daBee, 2022-03-16:
Testing this setup on a Raspberry Pi 2 Zero W, I get an error that it doesn't think it's running on a RPi.  I'm not sure if it's the recent release of this unit, so I thought I would post.  

```
RPi::GPIO.set_numbering :board
PINS = [5,7,11,13,14,16,18]

PINS.each do |i|
  RPi::GPIO.setup i, as: :input
  puts "#{ i } high? #{ RPi::GPIO.high?(i) }"
end
```

```
Traceback (most recent call last):
	3: from test.rb:22:in `<main>'
	2: from test.rb:22:in `each'
	1: from test.rb:23:in `block in <main>'
test.rb:23:in `setup': not running on a RPi (RuntimeError)
```

DeeJay, 2022-03-18:
To save the owner of this repo some investigation -

The current released version of Ben Croston's RPi.GPIO library is V0.7.1, and the corresponding Release Notes say "Added detection of Zero 2 W"
Ref: https://pypi.org/project/RPi.GPIO/0.7.1/

It seems reasonable to infer that an RPi Zero 2 W board, not available at the time, was NOT supported by v0.7.0 in July 2019. And hence will not be supported by this Ruby Gem.

Re-creating this Gem by incorporating the v0.7.1 sources might provide a solution for the owner of the Issue?

daBee, 2022-03-18:
As much as I would love to be able to write `C`, I do not.  I couldn't even guess as to how to incorporate this.