ClockVapor/rpi_gpio #28: Wrong ELF class on RPZ2

daBee, 2022-04-05:
Reinstalled `rpi_gpio 0.5.0` and it's reporting this issue on `Bundler.require`:

```
1: from /home/rich/.rbenv/versions/2.7.1/lib/ruby/gems/2.7.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio.rb:1:in `<top (required)>'
/home/rich/.rbenv/versions/2.7.1/lib/ruby/gems/2.7.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio.rb:1:in `require': /home/rich/.rbenv/versions/2.7.1/lib/ruby/gems/2.7.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio/rpi_gpio.so: wrong ELF class: ELFCLASS32 - /home/rich/.rbenv/versions/2.7.1/lib/ruby/gems/2.7.0/gems/rpi_gpio-0.5.0/lib/rpi_gpio/rpi_gpio.so (LoadError)
```

I've closed `ssh` sessions, and same issue.  Nothing else reporting errors, but then I'm not doing much on it outside of trying to get pins working.

daBee, 2022-08-26:
This gem is dead isn't it.

jongross, 2023-05-22:
Well I just ran into the same problem, and if I clone this and try and build it by hand, it won't even compile:

`~/rpi_gpio/ext/rpi_gpio $ ruby extconf.rb
creating Makefile
~/rpi_gpio/ext/rpi_gpio $ make
compiling c_gpio.c
c_gpio.c: In function ‘setup’:
c_gpio.c:134:9: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  134 |     if ((uint32_t)gpio_mem % PAGE_SIZE)
      |         ^
c_gpio.c:135:34: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  135 |         gpio_mem += PAGE_SIZE - ((uint32_t)gpio_mem % PAGE_SIZE);
      |                                  ^
c_gpio.c: At top level:
cc1: note: unrecognized command-line option ‘-Wno-self-assign’ may have been intended to silence earlier diagnostics
cc1: note: unrecognized command-line option ‘-Wno-parentheses-equality’ may have been intended to silence earlier diagnostics
cc1: note: unrecognized command-line option ‘-Wno-constant-logical-operand’ may have been intended to silence earlier diagnostics
compiling common.c
compiling cpuinfo.c
cpuinfo.c: In function ‘get_rpi_info’:
cpuinfo.c:86:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   86 |                case '0': info->type = "Model A"; info->p1_revision = 2; break;
      |                                     ^
cpuinfo.c:87:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   87 |                case '1': info->type = "Model B"; info->p1_revision = 2; break;
      |                                     ^
cpuinfo.c:88:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   88 |                case '2': info->type = "Model A+"; info->p1_revision = 3; break;
      |                                     ^
cpuinfo.c:89:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   89 |                case '3': info->type = "Model B+"; info->p1_revision = 3; break;
      |                                     ^
cpuinfo.c:90:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   90 |                case '4': info->type = "Pi 2 Model B"; info->p1_revision = 3; break;
      |                                     ^
cpuinfo.c:91:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   91 |                case '5': info->type = "Alpha"; info->p1_revision = 3; break;
      |                                     ^
cpuinfo.c:92:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   92 |                case '6': info->type = "Compute Module 1"; info->p1_revision = 0; break;
      |                                     ^
cpuinfo.c:93:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   93 |                case '8': info->type = "Pi 3 Model B"; info->p1_revision = 3; break;
      |                                     ^
cpuinfo.c:94:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   94 |                case '9': info->type = "Zero"; info->p1_revision = 3; break;
      |                                     ^
cpuinfo.c:95:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   95 |                case 'a': info->type = "Compute Module 3"; info->p1_revision = 0; break;
      |                                     ^
cpuinfo.c:96:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   96 |                case 'c': info->type = "Zero W"; info->p1_revision = 3; break;
      |                                     ^
cpuinfo.c:97:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   97 |                case 'd': info->type = "Pi 3 Model B+"; info->p1_revision = 3; break;
      |                                     ^
cpuinfo.c:98:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   98 |                case 'e': info->type = "Pi 3 Model A+"; info->p1_revision = 3; break;
      |                                     ^
cpuinfo.c:99:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   99 |                default : info->type = "Unknown"; info->p1_revision = 3; break;
      |                                     ^
cpuinfo.c:103:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  103 |                case '0': info->type = "Compute Module 3+"; info->p1_revision = 0; break;
      |                                     ^
cpuinfo.c:104:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  104 |                case '1': info->type = "Pi 4 Model B"; info->p1_revision = 3; break;
      |                                     ^
cpuinfo.c:105:37: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  105 |                default : info->type = "Unknown"; info->p1_revision = 3; break;
[... 290 lines cut ...]
[... cut ...]
/usr/bin/ld: rpi_gpio.o:/home/pi/rpi_gpio/ext/rpi_gpio/common.h:39: multiple definition of `pin_to_gpio_rev2'; common.o:/home/pi/rpi_gpio/ext/rpi_gpio/common.h:39: first defined here
/usr/bin/ld: rpi_gpio.o:/home/pi/rpi_gpio/ext/rpi_gpio/common.h:38: multiple definition of `pin_to_gpio_rev1'; common.o:/home/pi/rpi_gpio/ext/rpi_gpio/common.h:38: first defined here
/usr/bin/ld: rpi_gpio.o:/home/pi/rpi_gpio/ext/rpi_gpio/common.h:37: multiple definition of `gpio_mode'; common.o:/home/pi/rpi_gpio/ext/rpi_gpio/common.h:37: first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:262: rpi_gpio.so] Error 1`